OpenC3/cosmos #1458: Add ability to log selected telemetry value

321github123, 2024-08-07:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
We would like the ability to log a telemetry value over time

**Describe the solution you'd like**
User can select a telemetry item and a time (either received time or packet time).  UI would show the telemetry values (in text) over time.

User can add multiple telemetry item (in different tabs)

User can stop the logging.

User can configure logging frequency

User can export the log to file.

**Additional context**

[image]

[image]

[image]

sunws4me, 2024-08-07:
single tab could include multiple telemetry points as well.

jmthomas, 2024-08-08:
This would probably be best implemented as a Data Viewer widget. It allows you to output an arbitrary number of items but currently it is printing out the entire packet. We would just have to create a new widget to display an individual item with the associated timestamp

jmthomas, 2024-08-09:
Note this type of DataVIewer widget would also work well for JPL FPrime events which are best displayed in a time series.

321github123, 2024-09-16:
If possible, it would be great if this new data viewer widget had the following features

* user can add arbitrary number of items -- a single text line would display selected items from a single packet
* user can filter on a selected item
* user can freeze/unfreeze the flow of displayed data
* view persists if the user changes views and comes back

321github123, 2024-09-16:
@jmthomas Any chance you might have a rough estimate for when your team might release this future?  A ballpark figure, if available, would help our team's planning greatly.

321github123, 2024-09-16:
Another feature request would be the ability add items from different packets.  Each line of the output would be values of a single received packet e.g. if user chose packet_A.item_1, packet_A,item_2, and packet_B.item_1, the output would look like the following

```
time_1 packet_A.item_1 packet_A.item_2
time_2                                                     packet_B.item_1
time_3 packet_A.item_1 packet_A.item_2
time_4 packet_A.item_1 packet_A.item_2
```
There could also be an option (a toggle?) to propagate forward other packets values e.g.

time_1 packet_A.item_1 packet_A.item_2
time_2 (same as above) (same as above)  packet_B.item_1
time_3 packet_A.item_1 packet_A.item_2 (same as above) 
time_4 packet_A.item_1 packet_A.item_2 (same as above) 
```

jmthomas, 2024-09-17:
@321github123 this might not make the September release but I plan on having it done by the October release. It's a little more involved than I thought because DataViewer is currently Packet oriented and this is all item oriented. It's all possible but a little more frontend work than I thought. :-)

321github123, 2024-09-17:
Very helpful - thanks much!

PR #1593 (2024-10-01): Dv items
closes #1458
closed 2024-10-01 by jmthomas