OpenC3/cosmos #1460: Multiple targets from single target folder

mkalin2, 2024-08-07:
I have two of the same oscilloscope connected to COSMOS. I think I am using `TARGET` as described in the documentation to create multiple targets from a single folder. However, I get the following error when I send a command to an oscilloscope:
```
OSCILLOSCOPE_1_INTERFACE: RuntimeError : Telemetry target 'OSCILLOSCOPE' does not exist
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.17.1/lib/openc3/interfaces/protocols/cmd_response_protocol.rb:63:in `packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.17.1/lib/openc3/interfaces/protocols/cmd_response_protocol.rb:63:in `read_packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.17.1/lib/openc3/interfaces/interface.rb:300:in `block (2 levels) in read'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.17.1/lib/openc3/interfaces/interface.rb:299:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.17.1/lib/openc3/interfaces/interface.rb:299:in `block in read'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.17.1/lib/openc3/interfaces/interface.rb:259:in `loop'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.17.1/lib/openc3/interfaces/interface.rb:259:in `read'
interface_microservice.rb:488:in `run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.17.1/lib/openc3/microservices/microservice.rb:55:in `run'
interface_microservice.rb:715:in `<main>'
```

Here is what my plugin.txt looks like:
```
TARGET OSCILLOSCOPE OSCILLOSCOPE_1

INTERFACE OSCILLOSCOPE_1_INTERFACE tcpip_client_interface.rb xxx.xxx.xxx.xxx xxx xxx 10.0 None
  MAP_TARGET OSCILLOSCOPE_1
  PROTOCOL WRITE LengthProtocol 32 32 8 1 BIG_ENDIAN 0 nil nil TRUE
  PROTOCOL READ LengthProtocol 32 32 8 1 BIG_ENDIAN 8 nil nil TRUE
  PROTOCOL READ_WRITE CmdResponseProtocol
```

And here is what the command looks like:
```
COMMAND OSCILLOSCOPE GET_ID BIG_ENDIAN
  APPEND_PARAMETER OPERATION 8 BLOCK 0x81
  APPEND_PARAMETER HEADER_VERSION 8 UINT 1 1 1
  APPEND_PARAMETER SEQUENCE_NUMBER 8 UINT 1 1 1
  APPEND_PARAMETER SPARE 8 UINT 0 0 0
  APPEND_PARAMETER CMD_LENGTH 32 UINT 0 0 0
  APPEND_ID_PARAMETER CMD 40 STRING "*IDN?"
  RESPONSE OSCILLOSCOPE ID_RESP
```

It seems like this is a problem with the CmdResponseProtocol. The response packet target is not being updated from `OSCILLOSCOPE` to `OSCILLOSCOPE_1` or `OSCILLOSCOPE_2` as I would expect. Is this a bug or do I need to do something different?

jmthomas, 2024-08-07:
You need to use ERB in your command / telemetry definitions for the target renaming to work. Try changing your definitions to the following:
```
COMMAND <%= target_name %> GET_ID BIG_ENDIAN
```

mkalin2, 2024-08-08:
That seems to have worked!

closed 2024-08-08 by jmthomas