OpenC3/cosmos #1461: Wait check occassionally waits longer than timeout

321github123, 2024-08-07:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
The wait check waits longer than specified timeout

**To Reproduce**
This appears to be intermittent behavior

**Expected behavior**
Wait check waits correct amount of time

**Screenshots**

[image]

**Environment (please complete the following information):**

- OS: Ubuntu 20.04
- OpenC3 COSMOS Version 5.15.1
- Browser: Chrome

jmthomas, 2024-08-07:
The log says you waited 3378.415 seconds. Did your computer go to sleep and wake back up? Were the containers stopped?

sunws4me, 2024-08-08:
yes, everything seemed nominal.  I'll run a simple cmd loop script to send a noop every 10 seconds or so.  its happened twice to me and once to the other team.  no real pattern yet.

jmthomas, 2024-08-08:
I see it was a Python script. 3378.415 seconds is almost 1hr. Was this script being monitored or was it being run in the background without a frontend attached?

I know my Macbook will sleep and stop my containers from running. What platform is this running on?

jmthomas, 2024-08-29:
@sunws4me Any additional info?

closed 2024-09-24 by jmthomas