OpenC3/cosmos #1463: Data extractor time parameters can either be received time or packet time

321github123, 2024-08-08:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
We would like to data extraction either on received time or packet time.

**Describe the solution you'd like**
User has a choice of received time or packet time when using data extractor. Perhaps a dropdown?

**Additional context**
Add any other context or screenshots about the feature request here.

jmthomas, 2024-08-08:
I'm not sure what you mean because Data Extractor simply extracts the fields requested. PACKET_TIMESECONDS and RECEIVED_TIMESECONDS are separate fields that can be requested in the Data Extractor output.

321github123, 2024-08-09:
Hi,  data extractor contains fields that allows users to filter based on apparently PACKET_TIME

[image]

We would also like the ability to filter on RECEIVED_TIME

ryan-pratt, 2025-08-04:
Added this to Tlm Grapher in https://github.com/OpenC3/cosmos/pull/2249 but not Data Extractor