OpenC3/cosmos #1464: Telemetry Grapher Sometimes can't graph semi-recent Data

tsjordan-eng, 2024-08-09:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
I can't seem to plot data between about 15 minutes ago and an hour ago. I'll be plotting live data, then the old data disappears. I'll try to adjust the plot settings with the pencil tool to get that time and increase the points and seconds plotted in the graph settings, but that data is just gone. Then about an hour later I'll graph that data and it's able to graph it.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I expect the grapher to show me all data in the timeframe I request.

**Environment (please complete the following information):**

- OS: RHEL8
- OpenC3 COSMOS Version  5.16.2
- Browser: Edge

jmthomas, 2024-08-13:
I'm not able to reproduce. This is a pretty strange error. Is the data the normal data or reduced data? Are you navigating away from TlmGrapher and then back? Were you adding or removing items? I've never seen TlmGrapher "lose" data on a live graph. 

If you can reproduce on the Demo with a set of steps we'd have a much easier time tracking it down.

jmthomas, 2024-08-14:
Have you changed the `TLM_LOG_CYCLE_TIME` or `TLM_LOG_CYCLE_SIZE` for this target?

jmthomas, 2024-08-14:
I asked about cycle time because it kind of feels like you're pulling from redis but not from the log files. The way our architecture works, once the cycle time or log file size has been met, the streams are written out to files in the S3 bucket. When you pull data our streaming API pulls from those files and then from redis to display data.

You could verify in Bucket Explorer the files with the desired time range. If they are there then you should be pulling the data in TlmGrapher.

tsjordan-eng, 2024-08-14:
I was thinking that, too. I'm probably trying to plot a time window where half the data is in redis and half is in the streaming data. Do the Data Extractor and Telemetry Grapher handle time windows that span a log cycle action?

I'm having trouble reproducing it, but I will check the Bucket Explorer next time.

jmthomas, 2024-08-14:
Yes Data Extractor uses the same APIs as Telemetry Grapher so you should see the same behavior

jmthomas, 2024-08-29:
@tsjordan-eng Any new info?

closed 2024-09-27 by jmthomas

tsjordan-eng, 2024-09-27:
No... I think  it had something to do with  how telemetry packet time jumping back and forth causing problems.