OpenC3/cosmos #1471: Restrict command parameters to configured states

321github123, 2024-08-13:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
For a command parameter that has a set of states (i.e. an enumeration), would it be possible to restrict the user to only sending a parameter that is one of the states?

**Describe the solution you'd like**
The user cannot set a value via MANUALLY_ENTERED that is not a valid state.

**Additional context**
[image]

Cosmos allows the manually entering of a non-valid state, e.g., 777, and the sending the command with this parameter.

jmthomas, 2024-08-13:
If we're talking just about Command Sender then I think that's reasonable. We'll always allow the user to override the command parameter in a script because that's really useful for edge case testing in the UUT software.

If we disable the MANUALLY_ENTERED field in Command Sender they would still be able to go into the script output and edit it manually to send whatever.

321github123, 2024-08-14:
Disabling the MANUALLY_ENTERED in Command Sender sounds like a good near term solution.  Could we add a flag that enforces the state enumeration?

ryanmelt, 2024-08-14:
We can add a RANGE_CHECK_STATES flag

jmthomas, 2024-08-14:
We already have variations on cmd: `cmd, cmd_no_range_check, cmd_no_hazardous_check, cmd_no_checks`. It seems like this should be part of the `cmd_no_range_check` category. If you send a command with a state enumeration using `cmd` it should fail unless you use `cmd_no_range_check`. This is a pretty big breaking change though so we couldn't implement it until COSMOS 6.

ryanmelt, 2024-08-14:
The default behavior can't change until COSMOS 6, but we could add the new feature flag before then.

PR #1661 (2024-10-26): cmd and cmd_raw raise on bad item states
closes #1471 

Wait until COSMOS 6 to merge this behavior change.
PR #1670 (2024-11-08): Command State Range Checking for COSMOS 6
Enforces only defined states unless using cmd_no_range_check
closes #1471
closed 2024-11-08 by jmthomas