OpenC3/cosmos #1478: Ability to view raw data received from the interface

321github123, 2024-08-14:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
We can see the raw packet data but not the raw data received from the interface (before any protocols are applied)

**Describe the solution you'd like**
Something similar to Data Viewer. Additional features might be able to play back old data or save data to file.

**Additional context**
Add any other context or screenshots about the feature request here.

jmthomas, 2024-08-15:
We have the ability to do raw logging via the [`LOG_STREAM`](https://docs.openc3.com/docs/configuration/plugins#log_stream) keyword on the interface. As the docs note, this is the raw stream and we don't put any headers on the data so it can't be interpreted by any of the COSMOS tools.

We also have scripting APIs to start this raw logging: [`start_raw_logging_interface(<NAME>)`](https://docs.openc3.com/docs/guides/scripting-api#start_raw_logging_interface).

I think what you're asking for is additional tool support for these raw streams to be able to view them like we do with View Raw in the CmdTlmServer. Is that correct?

321github123, 2024-08-15:
_I think what you're asking for is additional tool support for these raw streams to be able to view them like we do with View Raw in the CmdTlmServer. Is that correct?_  

Yes that's right

ryanmelt, 2024-08-16:
That's not the right file (its a raw packet log file).  stream_logs will show up in BucketExplorer under DEFAULT/stream_logs

They will write out every 10 minutes or if you do a start, wait, stop with:
start_raw_logging_interface(<NAME>)
stop_raw_logging_interface(<NAME>)

They will show up immediately.

Note the double meaning of "raw logging".   In COSMOS 6 these APIs should be renamed to start_stream_logging_interface, stop_stream_logging_interface.

321github123, 2024-08-16:
I tried running the following script

```

print('hi')

start_raw_logging_interface("inst")

wait(60)

stop_raw_logging_interface("inst")

print('done')
```

but am not seeing "stream_logs" in bucket explorer (under log bucket)

[image]

ryanmelt, 2024-08-16:
This doesn't work with our simulated interface because no data is actually flowing in.   
If you connect the example interface and then use a similar script it will work.

connect_interface("EXAMPLE_INT")
start_raw_logging_interface("EXAMPLE_INT")
wait(60)
stop_raw_logging_interface("EXAMPLE_INT")