OpenC3/cosmos #1484: Add Packet Time to top of Packet Viewer

321github123, 2024-08-17:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
We would like to see the packet time of each packet (spacecraft time not received time)

**Describe the solution you'd like**
Perhaps add a field near the top of the viewer?

[image]

**Additional context**
Add any other context or screenshots about the feature request here.

jmthomas, 2024-08-17:
PACKET_TIME is already included in the Derived items

321github123, 2024-08-19:
Would it be possible to pin a specific item then?

jmthomas, 2024-08-20:
We could add a little pin icon on mouseover of a row and it would anchor the item to the top of the list. Pinning items would not be affected by the search so you could pin `PACKET_TIME` and then search for `TEMP` and it would show `PACKET_TIME` at the top along with `TEMP1`, `TEMP2`, etc.

321github123, 2024-08-20:
The pin icon would be a nice enhancement!

PR #1621 (2024-10-12): Pv pin
closes #1484
closed 2024-10-12 by ryanmelt