OpenC3/cosmos #1485: Can a Python script determine if it has been loaded with load_utility?

jacobian-m, 2024-08-17:
Is there a built-in or OpenC3-recommended way for a Python script to check whether it has been loaded through `load_utility()`?

(Context) I need to incorporate the equivalent to an `if __name__ == '__main__'` guard in Script Runner.  However, in Script Runner, `__name__` is `running_script` whether the script has been started directly or called through load_utility.  I can't `import` the code because it needs to be instrumented.  I can get around this, but a built-in COSMOS solution would be better if it exists already.

Thanks for your work on this project and any help you can provide.

jmthomas, 2024-08-24:
You can check for `RunningScript.instance`

closed 2024-08-24 by jmthomas

mkalin2, 2024-12-09:
That seems to work! Thanks!