OpenC3/cosmos #1489: TemplateAccessor Error when adding second SCPI command in single packet

mkalin2, 2024-08-20:
Here is my relevant plugin.txt:
```
TARGET POWER_SUPPLY POWER_SUPPLY

INTERFACE POWER_SUPPLY_INTERFACE tcpip_client_interface.rb xxx.xxx.xxx.xxx xxxx xxxx 10.0 None TERMINATED 0x0A 0x0A
  MAP_TARGET POWER_SUPPLY
  PROTOCOL READ_WRITE CmdResponseProtocol
```

The following command works as expected. It sets the voltage on CH1 of my power supply:
```
COMMAND POWER_SUPPLY SET_CH1_VOLTAGE BIG_ENDIAN "Set channel 1 voltage"
  ACCESSOR TemplateAccessor
  TEMPLATE "VOLT <VOLTAGE>, (@1)"
  APPEND_PARAMETER VOLTAGE 32 FLOAT 0 35 0
```

However, if I add a VOLT? query command and a response packet to the above command, I get an error when I try to send the command. I am doing this so that when I set a channel to a certain voltage, I can update a telemetry item with the channel's voltage set point.
```
COMMAND POWER_SUPPLY SET_CH1_VOLTAGE BIG_ENDIAN "Set channel 1 voltage"
  ACCESSOR TemplateAccessor
  TEMPLATE "VOLT <VOLTAGE>, (@1); VOLT? (@1)"
  APPEND_PARAMETER VOLTAGE 32 FLOAT 0 35 0
  RESPONSE POWER_SUPPLY CH1_VOLTAGE_SET_RESPONSE
```

**_This new command works_** in the sense that it correctly sends the command out of the interface, and I correctly receive and identify the response packet from the power supply. However, every time I send the command I get the following error in the command sender and in the CmdTlm log:
```
POWER_SUPPLY_INTERFACE: RuntimeError : Unexpected number of items found in buffer: 0, Expected: 1
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/accessors/template_accessor.rb:66:in `read_item'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/packets/packet.rb:583:in `read_item'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/packets/packet.rb:583:in `read_item'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/topics/command_decom_topic.rb:37:in `block in write_packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/topics/command_decom_topic.rb:36:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/topics/command_decom_topic.rb:36:in `write_packet'
interface_microservice.rb:208:in `block (2 levels) in run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/open_telemetry.rb:41:in `with_context'
interface_microservice.rb:71:in `block in run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/topics/interface_topic.rb:41:in `block in receive_commands'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:184:in `block (3 levels) in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:182:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:182:in `block (2 levels) in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:181:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:181:in `block in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:110:in `block (2 levels) in with'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:109:in `handle_interrupt'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:109:in `block in with'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:106:in `handle_interrupt'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:106:in `with'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:61:in `with'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:177:in `read_topics'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:89:in `public_send'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/utilities/store_autoload.rb:89:in `method_missing'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/topics/topic.rb:29:in `public_send'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/topics/topic.rb:29:in `method_missing'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.16.2/lib/openc3/topics/interface_topic.rb:40:in `receive_commands'
interface_microservice.rb:70:in `run'
interface_microservice.rb:54:in `block in start'
```

ryanmelt, 2024-08-20:
I think the problem is in your RESPONSE telemetry packet (which was not provided)

mkalin2, 2024-08-21:
I still get the error if I comment out the RESPONSE line, but here is how the response packet is defined:

```
TELEMETRY POWER_SUPPLY CH1_VOLTAGE_SET_RESPONSE BIG_ENDIAN "Channel 1 voltage set point"
  ACCESSOR TemplateAccessor
  TEMPLATE "<CH1_VOLTAGE_SET_POINT>"
  APPEND_ITEM CH1_VOLTAGE_SET_POINT 32 FLOAT
    FORMAT_STRING "%0.3f"
    UNITS "Volts" "V"
```

I have other telemetry packets that are defined just like this one, and they work fine.

jmthomas, 2024-08-22:
Can you post the exact response you're getting back from the hardware? We scan the template string `VOLT <VOLTAGE>, (@1); VOLT? (@1)` to look for the items and we're currently not escaping the `?` which may be breaking it.

mkalin2, 2024-08-22:
This is what is being sent:
```
Address   Data                                             Ascii
---------------------------------------------------------------------------
00000000: 56 4f 4c 54 20 35 2c 20 28 40 31 29 3b 20 56 4f  VOLT 5, (@1); VO
00000010: 4c 54 3f 20 28 40 31 29                           LT? (@1)   
```
And this is the response:
```
Address   Data                                             Ascii
---------------------------------------------------------------------------
00000000: 2b 35 2e 30 30 30 30 30 30 45 2b 30 30            +5.000000E+00   
```

jmthomas, 2024-08-23:
@mkalin2 thanks for reporting this. I have a fix that will be in the next release. In the meantime you'll have to separate the command / request into two commands.

closed 2024-08-23 by jmthomas

mkalin2, 2024-09-27:
Any status on when this new release will be?

ryanmelt, 2024-09-27:
Should be by next Tuesday.

PR #1495 (2024-08-23): Better escaping in TemplateAccessor
closes #1489