OpenC3/cosmos #1492: Data Backup

jmbell1989, 2024-08-22:
I have had multiple issue with drive space lately and it seems that docker has been eating up space in my app data directory. I am working on removing old docker container but don't want to lose my openc3 log data. Is there an easy way to backup this data and recover it at a later time? Any help with this would be appreciated.

Thanks!

jmthomas, 2024-08-22:
[This Docker article](https://www.docker.com/blog/back-up-and-share-docker-volumes-with-this-extension/) talks about the various options including the Volumes Backup & Share extension to backup a volume.

You can also do a [`docker system prune`](https://docs.docker.com/reference/cli/docker/system/prune/) to cleanup old docker images but not the volumes. If you've been developing COSMOS and building your own docker containers this can cleanup a significant amount of disk space.

jmthomas, 2024-09-06:
We need to document `docker system prune` as well as how to backup the minio volume.

jmthomas, 2026-09-02:
Closed by the following documentation: https://docs.openc3.com/docs/guides/backups

closed 2026-09-02 by jmthomas

EmilyRagan, 2026-09-03:
Closed by #3144