OpenC3/cosmos #1493: Enumerations On Telemetry/Command Definitions

lorenzo-gomez-windhover, 2024-08-22:
Hi there,

What is the recommended COSMOS way generating enumerated data inside Telemetry/Command ASCII files?

I understand that, in terms of encoding, enumerations are just integers, but I'm just wondering if COSMOS has a mechanism for presenting enumerations in a more user-friendly manner?

I looked up this topic in the documentation and github issues, but nothing came up. I'm new to COSMOS, so I apologize in advance if this seems like an obvious/silly question.

Thanks

lorenzo-gomez-windhover, 2024-08-22:
I should also note; I did find mention of enumerated types for XTCE. However given the fact that ASCII files are the first-class citizens in COSMOS, we decided to use the COSMOS ASCII format.

jmthomas, 2024-08-22:
We call these [STATE](https://docs.openc3.com/docs/configuration/command#state) in commands and [STATE](https://docs.openc3.com/docs/configuration/telemetry#state) in telemetry.

closed 2024-08-22 by jmthomas

lorenzo-gomez-windhover, 2024-08-22:
Thanks :+1: