OpenC3/cosmos #1499: System singleton access

BrunoBMoura, 2024-08-28:
On previous Cosmos versions, such as 4.5.X the access to the `System` singleton was granted to the whole code base, including scripts defined on the `targets` folder, on blocks such as:

```ruby
tlm_pkt = Cosmos::System.telemetry.packet("SOME_TARGET",  "SOME_TARGET_PACKET")
cmd_pkt = Cosmos::System.commands.packet("SOME_TARGET",  "SOME_OTHER_TARGET_PACKET")
```

However, on new OpenC3, when trying to perform the same call when executing  scripts on script runner, i get the following error: ```RuntimeError : System.instance parameters are required on first call```, which can be found on the `System` class implementation:https://github.com/OpenC3/cosmos/blob/a2c4e49c1fc140233c5124d20112d51f226c027c/openc3/lib/openc3/system/system.rb#L131

Is there a way to correctly make the `System` singleton accessible for scripts stored inside the `targets` folder on OpenC3?

ryanmelt, 2024-08-28:
If there is something specific, leave a comment.  Otherwise closing.

closed 2024-08-28 by ryanmelt

BrunoBMoura, 2024-08-28:
My use case is to simulate data on the cFS software bus as both commands and telemetries

It would be nice to have a similar approach for generating runtime data as the `build_cmd` API call, but also for telemetries, something like a `build_tlm` function.

Can this behavior be achieved using other API calls already provided?

ryanmelt, 2024-08-28:
inject_tlm is similar for telemetry (except it sends the telemetry packet through the system)