OpenC3/cosmos #1500: Cmd params not sending properly in cmd()

Lakshmi5785-dev, 2024-08-28:
I have been trying to upload data into by board.Here is my cmd method look like

cmd_name = "UPLOAD_DATA"
    cmd_params = {
      "PARAM1": 1,
      "PARAM2": 0,
      "PARAM3": "sldfjldsjfl"
    }
cmd(target_name, cmd_name, cmd_params)

In ruby When I send cmd it takes the last parametere DATA_BLOCK as hex value but in openc3 when I tried it is taking as a string.

When I read the documentation it says that "alue of the command parameter. Values are automatically converted to the appropriate type." -> What does this mean?

How can I send only the last parameter as hex without converting it manually?

ryanmelt, 2024-08-28:
"In ruby When I send cmd it takes the last parametere DATA_BLOCK as hex value but in openc3 when I tried it is taking as a string."

What do you mean by "ruby" and "openc3"?  In what context is it working like you expect, vs not?

jmthomas, 2024-08-28:
You're using the correct syntax: `cmd(target_name, cmd_name, cmd_params)`. So you just need to ensure your cmd_params are sending binary data. Here's an example of raw bytes: `cmd_params = { "DATA_BLOCK": "\x00\x01\x02\x03" }`.

jmthomas, 2024-09-05:
@Lakshmi5785-dev Any additional info? Did you ensure your cmd_params is correct?

Lakshmi5785-dev, 2024-09-06:
Hey Thomas, No. 
I am closing this ticket for now. I need to debug more on this. Once I get more details will open another ticekt depending on the need. Thank you for your support.

closed 2024-09-06 by Lakshmi5785-dev