OpenC3/cosmos #1509: Horizontal scroll bar for Telemetry Grapher

321github123, 2024-09-03:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
If we build a screen that is wide, there does not appear to be way to scroll horizontally left and right.  We are currently zooming out to see the entire screen.

**Describe the solution you'd like**
If this screen is wide, a horizontal scroll bar becomes available.

**Additional context**
Add any other context or screenshots about the feature request here.

jmthomas, 2024-10-18:
@321github123 TlmGrapher automatically scales the graph according to the display size. I'm not sure what you're looking for here.

ryanmelt, 2024-10-18:
I think TlmViewer was meant.

321github123, 2024-10-21:
Yes sorry I meant TlmViewer  -- thanks for the correction Ryan!  When we have many screens side by side, we are not able to scroll horizontally

PR #1639 (2024-10-24): Allow horizontal scrolling in apps
closes #1509
closed 2024-10-24 by ryanmelt