OpenC3/cosmos #1517: RSpec Tests Failing in GitHub Actions

stephen-ritter, 2024-09-05:
The GitHub actions do no appear to be flagging failing rpsec tests correctly. The rspec tests are failing under the `Run unit tests without extensions`, and you can see it here (I just chose the latest action against main):
https://github.com/OpenC3/cosmos/actions/runs/10641139981/job/29501905102

I did some tests locally and ran `rspec --bisect` against these tests and it appears these three tests are in conflict with another test. Here is my output:
```
Bisect started using options: "--tag=no_ext"
Running suite to find failures... (1.64 seconds)
Starting bisect with 3 failing examples and 488 non-failing examples.
Checking that failure(s) are order-dependent... failure appears to be order-dependent

Round 1: bisecting over non-failing examples 1-488 . ignoring examples 1-244 (0.72524 seconds)
Round 2: bisecting over non-failing examples 245-488 .. ignoring examples 367-488 (1.17 seconds)
Round 3: bisecting over non-failing examples 245-366 . ignoring examples 245-305 (0.4793 seconds)
Round 4: bisecting over non-failing examples 306-366 . ignoring examples 306-336 (0.45383 seconds)
Round 5: bisecting over non-failing examples 337-366 . ignoring examples 337-351 (0.42755 seconds)
Round 6: bisecting over non-failing examples 352-366 .. ignoring examples 360-366 (0.92483 seconds)
Round 7: bisecting over non-failing examples 352-359 .. ignoring examples 356-359 (0.93596 seconds)
Round 8: bisecting over non-failing examples 352-355 . ignoring examples 352-353 (0.44563 seconds)
Round 9: bisecting over non-failing examples 354-355 . ignoring example 354 (0.42457 seconds)
Bisect complete! Reduced necessary non-failing examples from 488 to 1 in 6.46 seconds.

The minimal reproduction command is:
  rspec './spec/config/config_parser_spec.rb[1:1:19]' './spec/packets/packet_config_spec.rb[1:1:25:1,1:1:25:2,1:1:25:3]' --tag=no_ext
  ```

jmthomas, 2024-09-05:
@stephen-ritter thanks for this report! We were overloading`exit()` in our spec_helper and not calling `super()` so we were eating the return code. I'm surprised this hasn't shown up in other places but it was an easy fix. I also fixed the example you specified by clearing the callback in the `config_parser_spec`.

stephen-ritter, 2024-09-05:
Thanks! I can confirm https://github.com/OpenC3/cosmos/pull/1518 fixed those tests as well.

PR #1518 (2024-09-26): Fix spec return code
closes #1517
closed 2024-09-26 by jmthomas