OpenC3/cosmos #1521: Add Node and NPM to base container image

camargo, 2024-09-06:
**Is your feature request related to a problem? Please describe.**

OpenC3 containers (e.g. ,the operator or cmd-tlm-api) do not have Node or NPM installed so you can't easily build plugins that require Node or NPM with those containers.

**Describe the solution you'd like**

Add `nodejs` and `npm` to the [base](https://github.com/OpenC3/cosmos/blob/main/openc3/Dockerfile) or [openc3-ruby](https://github.com/OpenC3/cosmos/blob/main/openc3-ruby/Dockerfile) image so we can use OpenC3 containers to build plugins that require Node and NPM.

jmthomas, 2024-09-24:
We already have a [Node Image](https://github.com/OpenC3/cosmos/blob/main/openc3-node/Dockerfile) as you've probably seen. You can use that image to build if you don't want to have a local installation of nodejs and npm.

From the current working directory that contains all your plugin files do the following:

Linux syntax:
```
docker run -it -v `pwd`:/openc3/local:z -w /openc3/local openc3inc/openc3-node sh
```
Windows syntax:
```
docker run -it -v %cd%:/openc3/local -w /openc3/local openc3inc/openc3-node sh
```

Then you should be able to do `yarn` and `rake build VERSION=1.0.0` to create the plugin.

We don't want to bloat our base images with nodejs as it adds a bunch of dependencies.

closed 2024-09-24 by jmthomas

camargo, 2024-09-25:
Thanks @jmthomas!