OpenC3/cosmos #1531: Query on open_files_dialog

Lakshmi5785-dev, 2024-09-10:
[image]
I tried to use the open_file_dialog and open_files_dialog as mentioned in the documentation. But both of them didn't work. For the open_files_dialog I am getting the list as empty whether I select a single file or multilple.
For the open_file_dialog Its throwing a syntax error.
[image]

jmthomas, 2024-09-11:
In COSMOS Open Source everything worked as expected.

In COSMOS Enterprise I was able to successfully open files but only as Admin (we will get that fixed in a future release). Can you identify the user you were using? Can you try as Admin?

Lakshmi5785-dev, 2024-09-12:
I am on a free license Thomas.

jmthomas, 2024-09-12:
I can not reproduce on COSMOS Open Source 5.18.0

PR #1544 (2024-09-19): Allow upload and delete to /tmp
closes #1531
closed 2024-09-19 by jmthomas