OpenC3/cosmos #1532: Python scripts ends after running `import datetime`

321github123, 2024-09-11:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
Script runner stops running python script ends after running the line "import datetime" 

**To Reproduce**

Steps to reproduce the behavior:

In Script Runner, run the following script

```
print('before import os')
import os
print('before import datetime')
import datetime
print('done!')

```

**Expected behavior**
We should see "done" in the console logs

**Screenshots**

[image]

**Environment (please complete the following information):**

- OS: Ubuntu 20.04
- OpenC3 COSMOS Version 5.17.1
- Browser: Chrome

jmthomas, 2024-09-11:
That is really weird. As a work around you can do this:
```
from datetime import datetime, timezone, timedelta
```

ryanmelt, 2024-09-12:
This is a side effect of our "magical imports" that make the COSMOS API available without having to import it.
Jason's answer is the current correct workaround.  I'll see if we can reduce the additional of unintentional imports like datetime which is causing this crash when it is imported as a module instead of a function.