OpenC3/cosmos #1562: Duplicate Interface

schuuust, 2024-09-23:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
I have one instance, MY_INT, running on COSMOS. It connects to another service running on my PC.  Sometimes COSMOS will display  a MY_INT_2 target as well as a MY_INT target, with telemetry only going through MY_INT_2. It has the same configurations as MY_INT_2, only the "my_target_name" variable changed

I can fix it by removing MY_INT_2 from COSMOS via the Admin panel, but it keeps reappearing. As a temporary workaround, I changed "my_target_name" to be be hardcoded as "MY_INT"  in the plugin.txt file. So far this stops the bug from recurring

**To Reproduce**
Unclear exactly when this bug occurs, I can't reliably replicate it. but I've noticed it in the following situations:
- Right after starting or rebooting my system (ubuntu 22.04) on and starting COSMOS and the MY_INT service
- After I've updated/otherwise interacted with MY_INT int COSMOS via automated web requests. This usually includes automating a "turn on" command via the script runner API or updating the gemfile 

**Expected behavior**
There is not a duplicate version of MY_INT in COSMOS

**Environment (please complete the following information):**

- Ubuntu 22.04
- OpenC3 COSMOS Version 5.15.2
- Browser: Firefox

ryanmelt, 2024-09-23:
We don't have any code that generates new names, so MY_INT_2 must have been named by you at some point.   Is this maybe a plugin you deleted, but then is being restored?

schuuust, 2024-09-23:
I don't believe I ever named "MY_INT_2". Its possible I accidentally introduced it via scripted requests at some point, but I don't see anything that could do that now.
Is there a way to prevent it from being restored?

ryanmelt, 2024-09-23:
Just trying to figure out what is happening.

Does it exist in your host's plugins/DEFAULT folder?

schuuust, 2024-09-23:
It did at one point, but I manually deleted it in an attempt to troubleshoot. I don't remember if that was before or after I last saw MY_INT_2

jmthomas, 2024-09-27:
@schuuust this is typically due to local files in the host plugins folder. If you want to disable local mode you can do so in the .env file by removing `OPENC3_LOCAL_MODE=1`.

closed 2024-09-27 by jmthomas