OpenC3/cosmos #1564: Packet decommutation rates matches rate of packets being sent

321github123, 2024-09-23:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
Packets of the same type are not being decommutated at a consistent rate.

**Describe the solution you'd like**
Packet decommutation rate matches rate of packets being sent.

**Additional context**
Below is a graph of packet decommutation speeds using Cosmos 5.18.1 Enterprise

[image]

As a comparison, a legacy T&C system is decommutaing with the following rates

[image]

ryanmelt, 2024-09-24:
Are you graphing received_time?   If so that is captured at the Interface, not at decom.  

Using garbage collected languages adds a lot of variability and I'm not sure if a rock solid line like you had is possible without using a compiled language.

321github123, 2024-09-24:
Yes we are graphing received time.