OpenC3/cosmos #1567: Writing to CSV file within a gem

hmaclachlan, 2024-09-24:
Hello. I am trying to create a log file (.csv) when a component is powered on/powered off to track burn time. I am wanting this log to be stored in my .gem, in the same folder that the script is in. The .gem that I am using only hosts procedures and screens (no target, interfaces, or cmds/tlm) as it is being used as a top level gem that hosts test scripts which interact with commands from other gems. My gem file structure is below:

- Script location that is appending to csv: MY_GEM/lib/utilities/power_on.rb
- Log file location that is being appended in power_on.rb: MY_GEM/lib/utilities/power_log.csv

This is how I am trying to append the log:

require 'csv'
require 'time'

#CSV.open('MY_GEM/lib/utilities/power_log_SAU_IMU.csv','a') do |csv|
#CSV.open('power_log_SAU_IMU.csv','a') do |csv|
  csv << ['Power on Time', Time.now.iso8601]
end

Originally, I ran the script with the first line of CSV.open uncommented, which resulted in an error for "no such file or directory". When I then ran the script with the second line of CSV.open uncommented, the script ran successfully but did not append the csv file. How do file paths work when working within the file structure of a gem? How do I get this log to append from another script and save it? Is there another way to do what I am trying to do here?

ryanmelt, 2024-09-24:
Build up your data, and then use put_target_file:
https://docs.openc3.com/docs/guides/scripting-api#put_target_file

You can retrieve the file with BucketExplorer or with get_target_file.

hmaclachlan, 2024-09-24:
How would I be able to append the csv file after its already created? Ie. I used put_target_file to create the file structure, but will need to append that .csv file every time an instance of a power on/off script is ran to track burn time of a component. The put_target_file creates a new file, and it seems that the get_target_file function just retrieves it. How do I modify the file?

tried this and ran successfully but did not append the file, again having issues with csv related functions:

require 'csv'
require 'time'

file = get_target_file("TRR_TEST_PROD/lib/utilities/power_log_SAU_test.csv", original: false)
CSV.open(file, 'a') do |csv|
  csv << [Time.now.strftime('%Y-%m-%d %H:%M:%S'),'ON']
end

ryanmelt, 2024-09-24:
You will need to put_target_file again at the end of your appends

hmaclachlan, 2024-09-25:
How do you denote that something is an append in put_target_file ? The only parameters are path and data. If the file already exists and I use put_target_file for the same file it overwrites the old file and makes a new one, removing any previous data.

How do you denote that the data parameter in this function is an append?

jmthomas, 2024-09-25:
You need to think about this differently. Initially you retrieve the data using `get_target_file`. Then you modify the data however you want. Finally you write the data back using `put_target_file`. If you do this same operation multiple times you're effectively appending or modifying in place. After the second `get_target_file` just append the data and write it back using `put_target_file`. You're never really working with the file itself ... you're working with the data in the file and then putting it back when you're done.

hmaclachlan, 2024-09-25:
Ok thanks. I get that you need to get_target_file, grab the data off of it to append it, and then use put_target_file to overwrite the original. I am having issues with getting the data back into the correct format and get an error. I have tried configuring the  data in different ways but continue getting an error. Any tips on how to configure this to not get a "broken pipe" error on the last line

require 'csv'
require 'time'

file = get_target_file("TRR_TEST_PROD/lib/utilities/power_log_test.csv", original: false)
csv_content = CSV.parse(file)
timestamp = Time.now.strftime('%Y-%m-%d %H:%M:%S')
status = 'ON'
log_instance = [timestamp, status]
csv_content << log_instance 
put_target_file("TRR_TEST_PROD/lib/utilities/power_log_test.csv", csv_content)

jmthomas, 2024-09-25:
You're trying to write an array to the file. Change the final line to this:
```
put_target_file("INST/data/power_log_test.csv", csv_content.to_csv)
```