OpenC3/cosmos #1575: load_utility not working in 5.18

Lakshmi5785-dev, 2024-09-26:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

Next search the issues here on Github. If you come up empty please submit an issue.
[image]

this works fine when we have a plain test, it fails when we add 
the below lines to the py file 

class TestSuite(Suite):
  def __init__(self):
      self.add_group(PING_TEST)

  def setup(self):
      # Run when Suite Setup button is pressed
      # Run before all groups when Suite Start is pressed
      pass

  def teardown(self):
      # Run when Suite Teardown button is pressed
      # Run after all groups when Suite Start is pressed
      pass

My scripts are working fine in 5.16 but are failing in 5.18

jmthomas, 2024-09-26:
The initialization routine in python is `def __init__(self)` not `def init(self)`. Can you try changing to `__init__` and see if that fixes it?

Lakshmi5785-dev, 2024-09-26:
I am sorry its a typo, I actually used __init__, but still i have the same error.
class TestSuite(Suite):
  def __init__(self):
      self.add_group(PING_TEST)

  def setup(self):
      # Run when Suite Setup button is pressed
      # Run before all groups when Suite Start is pressed
      pass

  def teardown(self):
      # Run when Suite Teardown button is pressed
      # Run after all groups when Suite Start is pressed
      pass

jmthomas, 2024-09-26:
Can you post your full script? The example isn't runnable.

Lakshmi5785-dev, 2024-09-26:
I tried ur my_test_suite.py , it works fine :(. I need to debug my script.
Thank you for the reply I am closing this ticket.

closed 2024-09-26 by Lakshmi5785-dev

jmthomas, 2024-09-26:
Also note that you can use File -> New Suite -> Python to generate the scaffolding for you.
[image]

Lakshmi5785-dev, 2024-09-26:
Yes, this is there in 5.16 as well right. By the way any idea when the above error occurs? I am unable to figure out the issue everything looks fine :(

ryanmelt, 2024-09-26:
You'll need to provide the complete script

Lakshmi5785-dev, 2024-09-27:
Hey Ryan, I am sorry I may not be able to share the full script. But below are few more details...
I am on v5.18.0
and this error is coming from the below file 
openc3/python/openc3/script/api_shared.py
[image]
May I know when this error raises?

jmthomas, 2024-09-27:
That code is overridden by [`running_script.py`](https://github.com/OpenC3/cosmos/blob/1f6f7c432d81ee83e7c4ac9c15ed9110e274cada/openc3-cosmos-script-runner-api/scripts/running_script.py#L1328). Thus if you're running outside of Script Runner you will get that error.

It looks like you're getting in during the suite analysis stage which happens before it's loaded in Script Runner. I tried a bunch of things but couldn't reproduce. I'd recommend starting with the File->New Suite->Python and then slowly copying your stuff into that structure to see if you can figure out what is causing it.

Lakshmi5785-dev, 2024-09-27:
Sure Thomas will try this out. This was working fine in 5.16. Looks line many new features added in 5.18
I was facing other issues as well. Will post them as well. All these suites worked fine in 5.16 :(

Lakshmi5785-dev, 2024-09-29:
Suite +load_utility combination might be the issue.
I am trying to debug from run_suite_analysis.py.
Unfortunately I am unable to use print , is there a way to log messages ?

Lakshmi5785-dev, 2024-09-29:
When I try to save the file in the script runner by keys ctrl+s it is trying to save the file in my local computer instead of saving in the script runner. I believe this is the new feature added.Is this causing the issue?
But when I tried my_script_suite.py in the demo INST2 it is just saving in the script runner container.

jmthomas, 2024-09-30:
We try to override the ctrl+s or command+s (Mac) to save the file. This seems to work well in Chromium browsers. What browser are you using?

In any case, the Save behavior should have no effect on what you're trying to do. Without a full script I don't think we'll be able to help you any further.