OpenC3/cosmos #1581: STATE Does Not Work For Telemetry Item

lorenzo-gomez-windhover, 2024-09-26:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
STATES for telemetry do not show up in web UI. They show up for commands in the Command Sender, but not for telemetry in PacketViewer.

**To Reproduce**
Steps to reproduce the behavior:

1. Get an instance of COSMOS up and running such as https://github.com/OpenC3/cosmos-project
2. Define a telemetry item like the following:
```
TELEMETRY CFS MM_HK_TLM_MID BIG_ENDIAN "Placeholder"
  #                NAME       BITS  TYPE    ID      DESCRIPTION
APPEND_ID_ITEM STREAM_ID 16 UINT 2183 "Stream ID" 
FORMAT_STRING "0x%04X"
APPEND_ITEM SEQUENCE 16 UINT "Packet Sequence" 
FORMAT_STRING "0x%04X"
APPEND_ITEM PKT_LEN 16 UINT "Length of the packet" 
APPEND_ITEM SECONDS 32 UINT "" 
UNITS Seconds sec
APPEND_ITEM SUBSECS 16 UINT "" 
UNITS Milliseconds ms
APPEND_ITEM SPARE2ALIGN 32 UINT "Spares array" 
APPEND_ITEM CmdCounter 8 UINT "Command Error Counter:" LITTLE_ENDIAN
APPEND_ITEM ErrCounter 8 UINT "Command Error Counter:" LITTLE_ENDIAN
APPEND_ITEM LastAction 8 UINT "Command Error Counter:" LITTLE_ENDIAN
APPEND_ITEM Padding 8 UINT "Command Error Counter:" LITTLE_ENDIAN
APPEND_ITEM MemType 32 UINT "Command Error Counter:" LITTLE_ENDIAN
 STATE MM_EEPROM 2
 STATE MM_MEM16 4
 STATE MM_MEM32 5
 STATE MM_MEM8 3
 STATE MM_NOMEMTYPE 0
 STATE MM_RAM 1

APPEND_ITEM DataValue 32 UINT "Command Error Counter:" LITTLE_ENDIAN
APPEND_ITEM _spare0 32 UINT "Command Error Counter:" LITTLE_ENDIAN
APPEND_ARRAY_ITEM FileName 8 INT 512 "" 
````

5. Go to Web UI's Packet Viewer and look for telemetry item.

**Expected behavior**
See the label as defined on the definition file instead of the integer value.
As you can see in the image below "MEMTYPE" shows "0" instead of "MM_NOMEMTYPE" as specified by STATE mapped to 0.

**Screenshots**

[image]

**Environment (please complete the following information):**

- OS: Pop!_OS 20.04 LTS
- OpenC3 COSMOS Version 5.17.1
- Firefox 130.0.1 (64-bit)

Thanks in advance
Lorenzo

lorenzo-gomez-windhover, 2024-09-26:
Or is this the way it is designed?

Does STATES only work on Screens when it comes to telemetry? I guess the PacketViewer is mean to be a "raw" view of data?

jmthomas, 2024-09-26:
Packet Viewer can show both. Check your View menu setting.

lorenzo-gomez-windhover, 2024-09-26:
@jmthomas Ah I see it. I feel silly now. Thanks so much for the quick response!

closed 2024-09-26 by lorenzo-gomez-windhover