OpenC3/cosmos #1584: Option for "Server Time" rather than browser time

jmthomas, 2024-09-27:
COSMOS time (including the Astro clock) is based on the local browser time and not the server time. Customer has asked for a way to force the Astro clock to the server time to better represent the actual RECEIVED_TIME that COSMOS is operating with. This would also eliminate the "Clock out of sync" warning that occurs when the local browser time is out of sync with the backend server.

Note we already collect the server time in TimeCheck.vue via `/openc3-api/time` so this information is available.

We should probably make this an option in the Admin next to the Timezone option (combine into 'Clock Settings' or something).

A side effect of basing time on server time is that the frontend time pickers will be still using local browser time which will make the defaults not as useful. I think that's ok for now and perhaps we warn the user when setting this mode that time choosers will need to be manually adjusted.

321github123, 2024-09-27:
Could we possibly make a backlog ticket for the front end time pickers to use server time?

321github123, 2024-09-27:
Also, would it be possible to set the admin clock setting via an api call? We would like a Kubernetes job to set all these settings for us in our deployed Cosmos