OpenC3/cosmos #1592: Is there a way to delete old command packets from redis database?

lorenzo-gomez-windhover, 2024-09-30:
Hi there,

Assume the following scenario:

1. Add a command called CFE_ES_StartApp:
```
COMMAND CFS CFE_ES_StartApp BIG_ENDIAN "Placeholder"
APPEND_ID_PARAMETER STREAM_ID 16 UINT 6150 6150 6150 "Packet Identification"
APPEND_PARAMETER SEQUENCE 16 UINT MIN_UINT16 MAX_UINT16 0xC000 "Sequence count" 
APPEND_PARAMETER PKT_LEN 16 UINT 113 113 113 "Packet length" 
APPEND_PARAMETER CMD_ID 8 UINT 4 4 4 "Command code" 
APPEND_PARAMETER CHECKSUM 8 UINT MIN_UINT8 MAX_UINT8 MIN_UINT8 "" 
APPEND_PARAMETER Application 160 STRING  "Command Error Counter:" 
APPEND_PARAMETER AppEntryPoint 160 STRING  "Command Error Counter:" 
APPEND_PARAMETER AppFileName 512 STRING  "Command Error Counter:" 
APPEND_PARAMETER StackSize 32 UINT MIN_UINT32 MAX_UINT32 MAX_UINT32 "Command Error Counter:"  "LITTLE_ENDIAN" 
APPEND_PARAMETER ExceptionAction 8 UINT MIN_UINT8 MAX_UINT8 MAX_UINT8 "Command Error Counter:"  "LITTLE_ENDIAN" 
  STATE PROC_RESTART 1
STATE RESTART_APP 0

APPEND_PARAMETER _spare0 8 UINT MIN_UINT8 MAX_UINT8 MAX_UINT8 "Command Error Counter:"  "LITTLE_ENDIAN" 
APPEND_PARAMETER Priority 16 UINT MIN_UINT16 MAX_UINT16 MAX_UINT16 "Command Error Counter:"  "LITTLE_ENDIAN" 
```

2. Rename the command to "CFE_ES_StartApp_New":

```
COMMAND CFS CFE_ES_StartApp_New BIG_ENDIAN "Placeholder"
APPEND_ID_PARAMETER STREAM_ID 16 UINT 6150 6150 6150 "Packet Identification"
APPEND_PARAMETER SEQUENCE 16 UINT MIN_UINT16 MAX_UINT16 0xC000 "Sequence count" 
APPEND_PARAMETER PKT_LEN 16 UINT 113 113 113 "Packet length" 
APPEND_PARAMETER CMD_ID 8 UINT 4 4 4 "Command code" 
APPEND_PARAMETER CHECKSUM 8 UINT MIN_UINT8 MAX_UINT8 MIN_UINT8 "" 
APPEND_PARAMETER Application 160 STRING  "Command Error Counter:" 
APPEND_PARAMETER AppEntryPoint 160 STRING  "Command Error Counter:" 
APPEND_PARAMETER AppFileName 512 STRING  "Command Error Counter:" 
APPEND_PARAMETER StackSize 32 UINT MIN_UINT32 MAX_UINT32 MAX_UINT32 "Command Error Counter:"  "LITTLE_ENDIAN" 
APPEND_PARAMETER ExceptionAction 8 UINT MIN_UINT8 MAX_UINT8 MAX_UINT8 "Command Error Counter:"  "LITTLE_ENDIAN" 
  STATE PROC_RESTART 1
STATE RESTART_APP 0

APPEND_PARAMETER _spare0 8 UINT MIN_UINT8 MAX_UINT8 MAX_UINT8 "Command Error Counter:"  "LITTLE_ENDIAN" 
APPEND_PARAMETER Priority 16 UINT MIN_UINT16 MAX_UINT16 MAX_UINT16 "Command Error Counter:"  "LITTLE_ENDIAN" 
```

3. Update/Remove and reupload CFS plugin. I tried both .

It seems that the old command still exists with the old name. Is this by design? If so, is there a way to wipe clean all of the commands/telemetry from the database so I don't have duplicated with different names?

Thanks in advance
Lorenzo

jmthomas, 2024-10-02:
Where does the old command exist with the old name? Command Sender? Data Extractor?

ryanmelt, 2024-10-03:
Also are you 100% sure both aren't in your definition files?

The situation you are describing is only possible if the uninstall process crashed.

lorenzo-gomez-windhover, 2024-10-03:
Apologies; This was operator error. Turns out I did have the command defined twice with two different names in two different definition files....

closed 2024-10-03 by lorenzo-gomez-windhover