OpenC3/cosmos #1618: Understanding SCOPE

lorenzo-gomez-windhover, 2024-10-04:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

Hello,

I'm trying to understand what SCOPES do in COSMOS. I understand that it is an Enterprise feature, but I'm trying to understand what it is exactly?

All the info I could find for it on https://openc3.com/enterprise suggests that it is a bucket for an "environment"? Does that mean that each scope has its own targets/screens/ command and telemetry dictionary? Or does the same target have multiple SCOPES? I guess I'm trying to understand the hierarchy there, if one could call it that. 

Hope the question makes sense.

Thanks
Lorenzo

ryanmelt, 2024-10-05:
Scopes are collections of different Interfaces and Targets, but running in the same COSMOS installation.  They provide an organizational and permission boundary.

"each scope has its own targets/screens/ command and telemetry dictionary" - Yes

lorenzo-gomez-windhover, 2024-10-07:
Thanks for the reply.

Forgive my lack of understanding;. I'm still trying to grok COSMOS architecture as a whole. Still trying to grasp the relationships between targets and plugins; can I have two SCOPES for the same instance of flight software? How does that work in terms of ports? Can two SCOPES be listening on the same port? In other words, can SCOPE just isolate/group data after it has been ingested into the system via UDP/Serial?

Hopefully the question makes sense.

jmthomas, 2024-10-07:
You can have multiple targets per scope (we have 4 in our Demo) but we recommend 1 to 1 for independence and maintainability.

You can have as many scopes as you want in Enterprise Edition. However this doesn't magically deconflict hostname, ports, or any other resource that is limited on a single machine. Scopes are all running on the same machine, they're just (relatively) independent of each other.

ryanmelt, 2024-10-07:
Jason meant you can have multiple targets per plugin.  (I corrected above).  Of course you can have many different targets in a scope.

You would almost never put the same flight software target in multiple scopes.  That would only be possible if the target could support multiple simultaneous connections.

Two scopes can only have interfaces listening on the same port if you are running in Kubernetes (then they will have different hostnames).

Scopes are meant to hold different targets.

lorenzo-gomez-windhover, 2024-10-07:
> They provide an organizational and permission boundary

Let me provide an example of what I understand so far:

So the idea is to be able to say, "Thermal" scope can only be viewed by Thermal Teams (I guess this could be accomplished with user accounts );  "Dev" scope can only be viewed by developers. Thermal scope  will have its own targets and so will Dev. Since we can have multiple targets for 1 plugin both scopes' data will be coming from the same vehicle (same UDP port), but operational speaking data/screens are separated so that different teams are not overwhelmed with all of the data from the vehicle.

Will this be the kind of thing once can accomplish with SCOPES in practice?

ryanmelt, 2024-10-07:
You should think of Scopes as seperate COSMOS installations.  If you could do it with seperate COSMOS installations, then you can do it with scopes (with the exception of reusing listening ports, unless you are on Kubernetes).

lorenzo-gomez-windhover, 2024-10-07:
Got it. Thanks a lot for your explanations (and patience) :+1:

closed 2024-10-07 by jmthomas