OpenC3/cosmos #1625: ROUTER: Add ability to specify a set of messages to route.

hockersmithjr, 2024-10-09:
Add ability to specify a set of messages to route in the plugin.txt and have the router only send/receive those messages.  This would reduce the need for custom code in order to route specific messages to an endpoint that only cares about certain messages.

jmthomas, 2024-10-10:
We already have an [ignore_packet_protocol](https://docs.openc3.com/docs/configuration/protocols#ignore-packet-protocol) that allows you to ignore certain "messages" (we call them packets).

ryanmelt, 2024-10-10:
You can attach protocols to routers (just like interfaces) to do things like filtering.  As Jason pointed out the ignore_packet_protocol should already do what you are asking for

hockersmithjr, 2024-10-10:
> External E-mail --- CAUTION: This email originated from outside GDMS. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Thanks.  I will give it a try.

closed 2024-10-10 by jmthomas