OpenC3/cosmos #1626: Is there a way to use ERB to define variable values from a higher-level variable?

GTI-Dakota, 2024-10-09:
Hello,

I am creating a Gem via a deployment script, and would like to be able to define a "prefix" variable that can be applied to other variables in the plugin.

For context: I have a system with one endpoint, and multiple subprocesses that I am splitting up into targets. I use one interface to communicate with the endpoint, and utilize the targets to designate commands, packets, screens, etc like a target would normally be used for. I need a "trickle-down" prefix variable because I will have N identical system configurations with different endpoints. I would like to keep things as general as possible in the gem creation script, and then designate/update a single variable on gem install. If this is not possible, that is no big deal, and I can work around it. 

Due to agile development of aforementioned system, cmd/tlm definitions change quickly, as well as the subprocesses and their names. Hard-coding VARIABLE lines into the deployment script are not possible.

auto-generated plugin example (non-functional):

> 
> VARIABLE prefix SYSTEM1_
> 
> VARIABLE subprocess1_target_name <%= prefix %>SUBPROCESS1
> VARIABLE subprocess2_target_name <%= prefix %>SUBPROCESS2
> VARIABLE subprocess3_target_name <%= prefix %>SUBPROCESS3
> VARIABLE subprocess4_target_name <%= prefix %>SUBPROCESS4
> VARIABLE subprocess5_target_name <%= prefix %>SUBPROCESS5

The issue here is that on gem install, the ERB kicks in after the user is prompted for the variable names, so the names are verbatim `<%= prefix %>SUBPROCESS1`, which wouldn't be a problem if the ERB applied the prefix to them, which it does not.

Thanks!

Dakota

jmthomas, 2024-10-10:
No unfortunately the ERB is a single pass. You'll have to apply your prefix lower down where the subprocess1_target_name is used:

```
TARGET <%= prefix %><%= subprocess1_target_name %>
```

GTI-Dakota, 2024-10-10:
Okay, makes sense. Thanks Jason!

closed 2024-10-10 by GTI-Dakota