OpenC3/cosmos #1629: Suggestion: allow option for termination characters inside terminated protocol packet

mkalin2, 2024-10-11:
I have an instrument that executes commands when it receives a newline character, so I am using the terminated protocol.

I had the desire to make my instrument do multiple things with a single COSMOS command, so I attended to insert a newline character into the middle of my packet which had two separate instrument commands in it so the packet definition looked like this:

`CMD 1 <newline> CMD 2` with a newline being added at the end by the terminated protocol.

However, I was unable to send this command due to the error checking inside the terminated protocol. It makes sense to have this warning by default, but perhaps there should be a plugin.txt parameter option to turn it off. There could be reasons that someone may want to have a termination character in their command. 

For now I have made a custom copy of the termination protocol for my plugin that has this error checking commented out.

mkalin2, 2024-10-11:
Also, let me know if this is going to break something. So far its working fine.

jmthomas, 2024-10-11:
I think that is a reasonable enhancement to the TerminatedProtocol. Probably a flag to the constructor: allow_termination_characters.