OpenC3/cosmos #1630: Implement a safety for leaving the script runner screen if a script is running

hmaclachlan, 2024-10-11:
Would it be possible to implement a safety feature or have a pop up appear on the screen if a user tries to exit the script runner while a script is in the middle of running? 

For example, if I were running a script that had a "wait" or a long "wait_check" and clicked the telemetry viewer while the script was running in the same tab that I was running the script in, it appears as though the script terminates. Would it be possible to have a feature that prevents the user from opening another view within the same tab (but allows another tab to be opened)?

jmthomas, 2024-10-11:
No the script is still running. Just go back to Script Runner and select Script -> Execution Status. Then you can connect to the existing script. We could potentially add an "Are you sure" when navigating away from a running script but otherwise we want people to be able to navigate.

hmaclachlan, 2024-10-11:
Thanks for the clarification on that - an "are you sure" would be nice, but as long as the script is still running the execution status is all I need.

ryanmelt, 2025-05-01:
Closing as returning to the script is now automatically attempted on the same tab.  No change needed.

closed 2025-05-01 by ryanmelt