OpenC3/cosmos #1631: Sometimes not being able to login to the localhost

MightyArty, 2024-10-16:
About a week or two ago we started to experience some weird behavior on new or existing setups with the cosmos.
In cases where we remove all the container's cache and data, and build them from scratch, when launching the localhost server usually we would have to type the username and the password and that's all.
But in the previous week or two, we experience that after inserting the username and the password and clicking on `set`, nothing happens and we stay at the login page forever, even after 10-15 retries.

The only thing that solved this issue for us is removing again all the container data and images from Docker, and running `openc3 start` to build the images from scratch.

If you have any suggestions on how we can overcome this behavior without running `openc3 start` each time, or if you encountered some similar cases before, we would be glad to receive some help.

Thanks,
Best Regards
Tom.

ryanmelt, 2024-10-16:
What version are you using?

mkalin2, 2024-10-17:
I was having this problem too. The password I used to always use was just 3 characters long, it looks like in a recent update there was a requirement added that the password needs to be longer than 8 characters. Unfortunately there isn't a pop-up error to tell you this. I had to F12 the page and look at the error console to figure this out. Is your password also short?

ryanmelt, 2024-10-17:
5.19.0 added an 8 character password requirement.   If there is no indication that the password is too short we'll get that fixed.

ryanmelt, 2024-10-17:
Old passwords shouldn't require a change though.

PR #1635 (2024-10-24): Catch initial password set errors
closes #1631
closed 2024-10-24 by ryanmelt