OpenC3/cosmos #1645: In the new version getting an error "COSMOS command parameters must be passed as strings: "validate"  "

MightyArty, 2024-10-20:
Hey, in the newest version `5.19.0` I've started to get an error when using the `cmd` command from openc3/script/commands.py that did not seen before.

When comparing to the older version of openc3 cosmos, noticed that you've added the validate parameter to the `def _cmd` function that is by default set to True.

So, when using the `cmd` command to pass the command and the command parameters as usual, we got a RuntimeError:

`RuntimeError: Unknown symbol keyword(s): validate. COSMOS command parameters must be passed as strings: "validate" => ...`

This is the whole Traceback of the error:

`  File "C:\Python\lib\site-packages\openc3\script\commands.py", line 32, in cmd
    return _cmd("cmd", "cmd_no_hazardous_check", *args, **kwargs)
  File "C:\Python\lib\site-packages\openc3\script\commands.py", line 227, in _cmd
    target_name, cmd_name, cmd_params = getattr(openc3.script.API_SERVER, cmd)(
  File "C:\Python\lib\site-packages\openc3\script\server_proxy.py", line 95, in method
    return getattr(self.json_drb, func)(*args, **kwargs)
  File "C:\Python\lib\site-packages\openc3\io\json_drb_object.py", line 107, in method
    return self.handle_response(response)
  File "C:\Python\lib\site-packages\openc3\io\json_drb_object.py", line 149, in handle_response
    raise error
RuntimeError: Unknown symbol keyword(s): validate. COSMOS command parameters must be passed as strings: "validate" => ...`

Tried to debug and solve it from my side, but did not succeed, unfortunately. 

Do I need to add the validate to the args that I'm sending to the `cmd` function (tried it and it did not solved the issue)?
Or is there any other thing that I can do to try and solve this issue on my own?

Thanks
Regards,
Tom.

ryanmelt, 2024-10-21:
This error is coming from the backend cmd tlm api.  Is that updated to version 5.19.0?

MightyArty, 2024-10-21:
> This error is coming from the backend cmd tlm api. Is that updated to version 5.19.0?

Yes, this is the updated 5.19.0 version

jmthomas, 2024-11-04:
@MightyArty has this been resolved? Is seemed like the cmd-tlm-api container was not up to date. We've now released COSMOS 5.20.0 so you can try again with that version.

jmthomas, 2024-12-14:
I assume this was resolved.

closed 2024-12-14 by jmthomas