OpenC3/cosmos #1672: Telemetry Grapher Does Not Graph Items That Contain "."

lorenzo-gomez-windhover, 2024-10-28:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
When I right-click a telemetry item that contains a "." character,  I get the following webpage:

[image]

**To Reproduce**
Steps to reproduce the behavior:

1. Go to Packet Viewer
2. Right-click on an item that contains "." such as `.PAYLOAD.UTILCPUPEAK`
3. Click "Graph"

**Expected behavior**
Telemetry Grapher should present the plot for the selected telemetry item. 

**Environment (please complete the following information):**

- OS: NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os

- OpenC3 COSMOS 5.17.1
- Browser: Firefox 131.0.2

ryanmelt, 2024-10-28:
It is highly likely that we will disallow the "." character in item names in the future, as this will likely be the syntax to pull from embedded objects.   I recommend renaming your items in this case.

lorenzo-gomez-windhover, 2024-10-28:
Thanks for the quick response!
@ryanmelt Is there a official/recommended way to name of naming fields of structs in COSMOS? I want to make sure we don't run into any issues like this in the future.

ryanmelt, 2024-10-28:
My recommendation is to avoid all special characters except single underscores _.  (note that double underscore__ is already not allowed).

However, we try not to be too controlling, as in the end it is just a string data type, so you could put anything in there.   Spaces will definitely give you a bad time though.

Also, we will probably continue to allow square brackets [ ] as several customers do that even though it is also our syntax for pulling from arrays so we have to add logic to check if it is an array item or just an item name that has brackets.   I find it confusing though so recommend avoiding that as well if you can.

jmthomas, 2024-11-01:
After talking to other customers I think we're going to proceed with supporting periods in item names. I'll update the PR to reflect this and we'll add whatever changes we need to TlmGrapher and other tools to handle it.

closed 2024-11-15 by jmthomas