OpenC3/cosmos #1673: How to consistently read telemetry items without having to send a SCPI query command.

hmaclachlan, 2024-10-28:
Hello, 

I have a device that is my target which operates via SCPI commands. I want to obtain consistent telemetry readings from this target. From my current plugin, cmd, and tlm.txt files, I am only able to receive the output from a query command as telemetry when the command is sent. Is there a way to configure the plugin.txt differently to allow the telemetry to be acquired consistently without having to send a command each time?

VARIABLE oscope_target_name OSCOPE

TARGET OSCOPE <%= oscope_target_name %>
INTERFACE <%= oscope_target_name %>_INT tcpip_client_interface.rb xxx.xxx.xxx.xxx xxxx xxxx 10.0 None TERMINATED 0x0A 0x0A
  MAP_TARGET <%= oscope_target_name %>
  PROTOCOL READ_WRITE CmdResponseProtocol

COMMAND OSCOPE GET_MEAS1 BIG_ENDIAN "Get Measurement 1"
  ACCESSOR TemplateAccessor
  TEMPLATE "MEASU:MEAS1:VAL?"
  RESPONSE OSCOPE GET_MEAS1_RESPONSE

TELEMETRY OSCOPE GET_MEAS1_RESPONSE BIG_ENDIAN "OSCOPE MEAS1 Telemetry"
  ACCESSOR TemplateAccessor
  TEMPLATE "<GET_MEAS1>"
  APPEND_ITEM GET_MEAS1 32 FLOAT "Measurement Value 1"
    FORMAT_STRING "%0.3f"

ryanmelt, 2024-10-28:
I don't think its documented, but all interfaces have an option called PERIODIC_CMD:

https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/interfaces/interface.rb#L502

Add this after your INTERFACE line in plugin.txt and it will periodically send the command in quotes.
LOG means log each command sent.  Generally you want to put DONT_LOG for periodic commands (they are still logged in our packet logs, but won't show in the CmdTlmServer log messages).
1.0 here is the period in seconds
then the string is the command to send like in ScriptRunner.

```OPTION PERIODIC_CMD LOG/DONT_LOG 1.0 "INST COLLECT with TYPE NORMAL"```

hmaclachlan, 2024-10-28:
@ryanmelt  Thanks for the help! One follow up question - is it possible to have multiple PERIODIC_CMD responses recorded in the same telemetry packet? The response packet, defined below, does not work, I believe because of the multiple TEMPLATE inputs.

TELEMETRY OSCOPE TLM BIG_ENDIAN "OSCOPE Telemetry"
  ACCESSOR TemplateAccessor
  TEMPLATE "<GET_MEAS1>,<GET_MEAS2>"
  APPEND_ITEM GET_MEAS1 32 FLOAT "Measurement Value 1"
    FORMAT_STRING "%0.3f"
  APPEND_ITEM GET_MEAS2 32 FLOAT "Measurement Value 2"
    FORMAT_STRING "%0.3f"

ryanmelt, 2024-10-28:
@hmaclachlan You'll need to combine the SCPI commands into one big SCPI command that queries everything, and then it should work.

hmaclachlan, 2024-10-28:
@ryanmelt How do you correctly configure/parse the separate telemetry outputs when you combine multiple SCPI commands? Getting the error (Decom error: RuntimeError : Unexpected number of items found in buffer: 0, Expected: 4) and the entire packet is reading null with this configuration:

VARIABLE oscope_target_name OSCOPE

TARGET OSCOPE <%= oscope_target_name %>
INTERFACE <%= oscope_target_name %>_INT tcpip_client_interface.rb xxx.xxx.xxx.xxx xxxx xxxx 10.0 None TERMINATED 0x0A 0x0A
  OPTION PERIODIC_CMD DONT_LOG 10.0 "OSCOPE GET_MEAS"
  MAP_TARGET <%= oscope_target_name %>
  PROTOCOL READ_WRITE CmdResponseProtocol

COMMAND OSCOPE GET_MEAS BIG_ENDIAN "Get Measurements"
  ACCESSOR TemplateAccessor
  TEMPLATE "MEASU:MEAS1:VAL?;MEASU:MEAS2:VAL?;MEASU:MEAS3:VAL?;MEASU:MEAS4:VAL?"
  RESPONSE OSCOPE GET_MEAS_RESPONSE

TELEMETRY OSCOPE GET_MEAS_RESPONSE BIG_ENDIAN "OSCOPE Telemetry"
  ACCESSOR TemplateAccessor
  TEMPLATE "<GET_MEAS1>;<GET_MEAS2>;<GET_MEAS3>;<GET_MEAS4>"
  APPEND_ITEM GET_MEAS1 32 FLOAT "Measurement Value 1"
    FORMAT_STRING "%0.3f"
  APPEND_ITEM GET_MEAS2 32 FLOAT "Measurement Value 2"
    FORMAT_STRING "%0.3f"
  APPEND_ITEM GET_MEAS3 32 FLOAT "Measurement Value 3"
    FORMAT_STRING "%0.3f"
  APPEND_ITEM GET_MEAS4 32 FLOAT "Measurement Value 4"
    FORMAT_STRING "%0.3f"

ryanmelt, 2024-10-28:
The TELEMETRY TEMPLATE has to be just right.   Take a look at the raw data if you can and look for spaces etc.  Those will have to be in the template.

Also I know we made a fix for this in 5.19.0.   If you're not on the version, you'll need it or 5.20.0 which we'll be releasing tomorrow.

https://github.com/OpenC3/cosmos/issues/1489

closed 2024-12-04 by hmaclachlan