OpenC3/cosmos #1697: How do I start a Python script from Ruby?

tsjordan-eng, 2024-11-10:
I have used the `runScript` call in the telemetry panels to start a script, but I can't find a way to do this in the Script Runner. It would be helpful because we have existing scripts and suites written in Ruby, but we want to take advantage of using Python scripts for their libraries.

ryanmelt, 2024-11-11:
# Spawns a script in ScriptRunner
script_run("INST2/procedures/checks.py")

tsjordan-eng, 2024-11-11:
Ah, that seems to work! I couldn't find it in the documentation, though I suspect it has the same arguments like passing environment variables and such. I'll check out the source code.

jmthomas, 2024-12-02:
Ensure we document all the APIs under `openc3/script/*`