OpenC3/cosmos #1715: Python import in a TestSuite

GabeBarfield, 2024-11-19:
I submitted a previous ticket on this topic which was closed and indicated we should use COSMOS latest (5.20.0). We ran into a myriad of issues with 5.20.0 and thus are still on 5.17.1. Again the issue is we moved our scripting to Python from Ruby. When we want to import a python file, to be used by the test suite we get "ModuleNotFoundError: No module named 'TGT'". 

To produce this issue I create a new plugin with:
**opt/openc3/cosmos-5.17.1/openc3.sh cli generate plugin tgt**
Plugin openc3-cosmos-tgt successfully generated!

and a new target with:
**/opt/openc3/cosmos-5.17.1/openc3.sh cli generate target tgt --python**
Target TGT successfully generated!

I update my plugin and target files (provided below)
[plugin.txt](https://github.com/user-attachments/files/17817811/plugin.txt)
[target.txt](https://github.com/user-attachments/files/17817815/target.txt)

I build the plugin and deploy. In script runner I edit procedure.py to have the following:
**class HellowWorld:
  def print_hello_world(self):
    print("Hello World!")**

I then create a New Test Suite from template and add the line:
**from TGT.procedures.procedure import ***

This is where I get the error from above. This is effectively a generated default plugin with a small edit and Python importing doesn't work as the user guide indicates.

Thanks.

Ubuntu 22.04.4
Cosmos 5.17.1

ryanmelt, 2024-11-19:
Please let us know what issues you ran into on 5.20.0.    Again we think this has already been resolved in a newer version.

GabeBarfield, 2024-11-19:
My previous issues were from updating my existing plugin, so to be safe I ran the steps above on 5.20.0 and I get the same issue.

GabeBarfield, 2024-11-19:
My previous issues had to do with updating the old plugin to python while also fighting this issue. I ran the same steps above changing 5.17.1 to 5.20.0 (the plugin gen did require an additional python tag, shown below) and I have the same result.

/opt/openc3/cosmos-5.20.0/openc3.sh cli generate plugin tgt --python
/opt/openc3/cosmos-5.20.0/openc3.sh cli generate target tgt --python
Update plugin.txt and target.txt files to what was previously attached
/opt/openc3/cosmos-5.20.0/openc3.sh cli rake build VERSION=1.0.0
Load plugin in COSMOS 5.20.0. 
Generate Test Suite and include:
from TGT.procedures.procedure import *
"ModuleNotFoundError: No module named 'TGT'".

jmthomas, 2024-12-03:
I was able to reproduce with Suites in Script Runner. I don't think this actually affects how the suite runs but it won't be able to to be parsed and thus generates this error dialog.

PR #1730 (2024-12-04): Py suite
closes #1715
closed 2024-12-04 by jmthomas