OpenC3/cosmos #1725: Way to update a TLM item based on sending a command

mkalin2, 2024-11-25:
Lets say I have a stepper motor driver, and I want to keep track of the total number of steps that I have commanded to the motor. Is there a way to have the 'motor move' command packet update a TLM packet that will accumulate the number of steps across the command sender and various scripts?

It seems like I could do this using a write conversion and a virtual TLM packet. I am unsure of how I to access the TLM packet from the conversion however. I am also unsure of how I could reset this TLM value without actually sending a command to my motor driver board...

Is this possible and if so am I on the right track?

jmthomas, 2024-11-25:
You can certainly do this with "fake" telemetry packets and by calling [`inject_tlm`](https://docs.openc3.com/docs/guides/scripting-api#inject_tlm) to update it. However I would recommend using the [Stash API](https://docs.openc3.com/docs/guides/scripting-api#stashing-data) which was created for just this use case. Store something internally and then retrieve it.

mkalin2, 2024-11-25:
I would like this to work even if the command is sent via the command sender. Will I be able to update the stashed value outside of a script?

mkalin2, 2024-11-25:
Like is there a way to implement a custom telemetry item like `RECIVED_COUNT` that gets updated based on things that happen in COSMOS rather than by info in the packet.

jmthomas, 2024-11-25:
You could create a command [VALIDATOR](https://docs.openc3.com/docs/configuration/command#validator) that does an `inject_tlm` like I mentioned above to save whatever values you want in an existing telemetry packet.