OpenC3/cosmos #1726: Issues with minio container when downgrading from 5.18.0 to 5.15.0

hmaclachlan, 2024-11-25:
**Describe the bug**
I need to downgrade my local COSMOS working environment to match the version of another environment in an external location. When I try to downgrade from 5.18.0 to 5.15.0, the minio container does not successfully start, which results in a bad gateway when trying to access localhost:2900. I followed these steps: 
[image]

I tried to downgrade in another project from 5.19 to 5.15 and that worked, but when trying to go from 5.18 to 5.15 I have issues. I duplicated this issue in two separate 5.18 projects. 

**To Reproduce**
Try to downgrade a 5.18 COSMOS project.

**Expected behavior**
Being able to access localhost:2900. 

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS: Windows 10
- OpenC3 COSMOS Version 5.18.0 
- Browser: Edge

jmthomas, 2024-11-25:
Downgrading is not necessarily supported between minor revisions ... we have migration scripts to upgrade but if you're planning on downgrading you should first save your container volumes first. We should really document this better.

ryanmelt, 2024-11-25:
We don't guarantee compatibility when downgrading.   Lots of times databases and things have changed with newer container versions.  

You'll probably need to recreate your volumes (note this will lose all your data).  That can be done with ./openc3.sh cleanup, and then openc3.sh run.

closed 2024-12-04 by hmaclachlan