OpenC3/cosmos #1727: Using tlm api in a Python protocol

thomasstjeangenevatech, 2024-11-26:
Hello,

I have a question that is similar to [this one](https://github.com/OpenC3/cosmos/issues/744), but for a python protocol. I am trying to access a telemetry item from a protocol so that I can use the previous value before I return a new value for the same telemetry item. However, no matter what method I try, I get this permission error: `redis.exceptions.NoPermissionError: No permissions to access a key`. I am including the method like so `from openc3.api.tlm_api import tlm`. Do I need to do this differently for it to work? Let me know if you need more info.

Thanks

ryanmelt, 2024-12-03:
What you are doing should work (though I don't recommend using APIs in protocols for performance reasons).  Have you tried on 5.20.0?

thomasstjeangenevatech, 2024-12-03:
I have not tried on 5.20.0, although I could if that would be helpful information to you (our production servers are all running 5.17.1 currently). If using the APIs is not recommended, is there another way to access telemetry from a protocol without them?

ryanmelt, 2024-12-03:
It would be good to know if this is something we have already fixed in 5.20.0.

It is better to keep the data locally in the protocol if possible rather than going across the network to Redis.  If the protocol can read the value from the packet itself and save the previous value that is better.   You can do that in the read_packet() method (you might have to identify the packet first).