OpenC3/cosmos #1732: How to exit a script if an error occurs in a parallel script

hmaclachlan, 2024-12-04:
I have two scripts that run in parallel. The parallel script is initiated from the parent script via script_run() and loops through various telemetry checks. I am trying to implement logic into both scripts that stops the parent script if an error occurs in the parallel script, but also allows the parent script to continue running when prompted by the user. I have tried doing this via using a file to define parallel script status. Example code below:

snippet from a parent script that checks parallel script status:
```
script_run('__TEMP__/parallel.rb')
wait(10)
loop do
  status = get_target_file("__TEMP__/parallel_script_status.txt")
  status = File.read(status)
  puts status
  if status == 'FAILED'
    exit
  end
end

puts 'User continuing script'
```
parallel script:
```
def tlm_validity(tlm_item)
  if tlm(tlm_item) == 'VALID'
    puts "#{tlm_item} IS VALID"
    return 1
  else
    puts "#{tlm_item} IS INVALID"
    return 0
  end
end

loop do
  tlm_1_status = tlm_validity("TARGET PACKET TLM1")
  tlm_2_status = tlm_validity("TARGET PACKET TLM2")
  tlm_3_status = tlm_validity("TARGET PACKET TLM3")
  if tlm_status_sum == 3
    put_target_file("__TEMP__/parallel_script_status.txt", "COMPLETED")
  else
    put_target_file("__TEMP__/parallel_script_status.txt", "FAILED")
    exit
  end
  wait(60)
end
```

The method above works for exiting the parent script when the parallel script fails. This still presents issues, however, as I need to be able to break this loop and continue the parent script upon user input. Crtl + C does not work in the Script Runner IDE to break the loop. How would I break this loop when the user desires to (after waiting for the test to run to completion). Also, is there a more streamlined way to do what I am trying to do here with checking script status?

*I have also tried this by using global variable in each script using $, ie `$parallel_script_status = 'FAILED'`, in place of the status file. I have found that when the global variable is defined in one script, it's definition is not carried over when referenced in another script. Is there something I am missing here in terms of using global variables in COSMOS?

ryanmelt, 2024-12-05:
Scripts run in their own processes so global variables are not shared between scripts.

Easier than using files like you are doing, would be to use the stash api to set a flag. 
https://docs.openc3.com/docs/guides/scripting-api#stashing-data

We don't currently have a UI method to break the loop like you want to.  I'll write a ticket to do that.  For now you'll have to set a flag somehow somewhere else (stash api would work that for that too).  You could have a third script that passes something to the stash api, and then your main script checks that to see if it needs to break.

jmthomas, 2024-12-14:
re #1734

closed 2024-12-14 by jmthomas