OpenC3/cosmos #1735: Script Suite Variable Fail/Error Pause Conditions

mkalin2, 2024-12-05:
I would like to have a script suite that will pause on errors like an interface not responding, but will not pause -- and simply fail the test -- for errors like from check_tolerance where a telemetry value is just out of range.

Is there a way to implement something like this currently? Could I do the tolerance checking myself and assert a failure at the end of the script without forcing the operator to intervene?

The end result I am looking for is to be able to run a script suite and know that if something is fundamentally broken it will pause and alert the operator, but otherwise allow the operator to assess the overall pass/fail of the test by looking at the script suite report after everything has run.

ryanmelt, 2024-12-05:
You can wrap check statements with exception catching (begin/rescue in ruby) (try/except in python) and then do whatever if the check fails (including do nothing)

mkalin2, 2024-12-05:
try/except does not seem to prevent check statements from pausing the script execution...

PR #1855 (2025-01-27): Improve python instrumentation
closes #1735 
closes #1406
closed 2025-01-27 by jmthomas