OpenC3/cosmos #1738: How to handle packets with varying data structure?

badrobit, 2024-12-06:
I have a series of CSP packets that we are processing. Inside those packets, in the payload, oftentimes, there is a "status" field, and based on that status, the rest of the packet definition can vary. 

e.g. 

Shared between all packets (same CSP node/port and ID field):  
```
field_1
status
field_2 
```

Then, if `status` is say, `0,` I could have another ten possible fields: 
```
field_3
...
field_12
```

so with the same CSP node/port/ID I can have two unique packet "structures" 

`status=0`
```
field_1
status (=0)
field_2
```

or `status != 0` 
```
field_1
status (!=0)
field_2
field_3
field_4
field_5
field_6
field_7
field_8
field_9
field_10
field_11
field_12
```

But if status is anything but `0` then I will only get the first 3 fields. I looked at the Array parameter type, and it almost works, but only if all the additional fields are the same data type (they are not in this case sadly)

ryanmelt, 2024-12-06:
Whenever you have a field that controls the contents of a packet, you almost always want that field to be part of the ID_ITEMS.  That way you can define different packets based on that value.

Because the status value covers a range of values for the second definition, you could potentially define the packets like this:

Telemetry definitions are processed in order of definition. So you can do something like this:

```
TELEMETRY TARGETNAME PACKET1 ...
   ...
   APPEND_ID_ITEM status ... 0 # Specifically look for zero
   ...

TELEMETRY TARGETNAME PACKET2
  ...
  APPEND_ITEM status ... # Not an id item so will catch anything else
  ...
```

Then in target.txt for your target you will need to add:
```
TLM_UNIQUE_ID_MODE
```

badrobit, 2024-12-06:
Thanks, that makes sense. Can I do the same for commands? also just to make sure I flush it out say I have three possible statuses can I do 

```
TELEMETRY TARGETNAME PACKET1 ...
   ...
   APPEND_ID_ITEM status ... 0 # Specifically look for zero
   ...

TELEMETRY TARGETNAME PACKET2
  ...
  APPEND_ID_ITEM status ... 1 # Specifically look for zero
  ...

TELEMETRY TARGETNAME PACKET3
  ...
  APPEND_ID_ITEM status ... 2 # Specifically look for zero
  ...

TELEMETRY TARGETNAME PACKET2
  ...
  APPEND_ITEM status ... # Not an id item so will catch anything else
  ...

```

just to triple check my understanding

ryanmelt, 2024-12-06:
Yes you can do the same with commands.  You'll need to set CMD_UNIQUE_ID_MODE in target.txt for them.

If you define all the possibilities you can eliminate the catchall packets (and the UNIQUE_ID_MODE options).

badrobit, 2024-12-06:
one last question in the same point of view (thank you so much for the rapid responses). 

I have a command I need to implement, which has again a fixed structure but then I can add "sections" to the command which are all chunks of the same structure

e.g. this is the static part: 
```
COMMAND TARGET COMMAND_NAME BIG_ENDIAN "<DESCRIPTION>" 
   APPEND_ID_PARAMETER COMMAND_ID
   APPEND_PARAMETER field_1 ...
   APPEND_PARAMETER field_2 ...
   APPEND_PARAMETER field_3 ...
```

Then the user can add anywhere between 1 and 60 sections which would look like: 
```
   APPEND_PARAMETER section_field1 UINT 32
   APPEND_PARAMETER section_field2 UINT 16
   APPEND_PARAMETER section_field3 UINT 16
```

badrobit, 2024-12-06:
When using the  `CMD_UNIQUE_ID_MODE` and `TLM_UNIQUE_ID_MODE` does it treat each `ID` parameter/item as if it has to be unique or if the combination of all ID items has to be unique? I have a CSP header defined and brought into the packet definitions using a `<%= render "_CUBE_SAT_PROTOCOL.cosmos", locals: {destination: 3, destination_port: 7} %>` in the command definition. 

What I am trying to accomplish is that with the `CSP_DESTINATION`, `CSP_DESTINATION_PORT`, and the `COMMAND_ID` field on a specific command, I could have a truly unique identifier across all the various command and telemetry files I am ingesting into the COSMOS instance. 

This is because CSP allows for the reuse of command_ids across various nodes/ports, so I can only uniquely identify something when taking the destination node/port/command_id into account. 

I could work around it by generating an internal hidden value for each command to store a unique ID I could generate as I render the files. Still, I was hoping the `COMMAND_UINQUE_ID_MODE` would make it so that across all commands for the defined target, the combination of all identified "ID" fields could be used as a unique identifier instead of managing that myself. This is mostly an attempt to keep the telemetry files we generate as close to the input definitions we are given.

closed 2024-12-14 by jmthomas