OpenC3/cosmos #1741: Cannot seem to open any tools in demo open source version

guy0nAbuffal0, 2024-12-06:
When I click the tools, it will refresh the page and the web address at the top will change, but I don't see any tools load.
I have tried two different browsers and verified that popup blocking is turned off.
Is there anything additional I need to do so these appear?

[image]

ryanmelt, 2024-12-07:
You appear to have built from source.  We are in the middle of a major change and the main branch is not guaranteed to be fully working at any given time.   

Please follow the readme to checkout cosmos-project and run that with our released 5.20.0 containers.

closed 2024-12-07 by ryanmelt

guy0nAbuffal0, 2024-12-09:
Thank you, when I instead cloned the cosmos-project and then rebuilt, it worked great!