OpenC3/cosmos #1745: Allow Configuring Cleanup Microservice for Non-Packet Logs

JoshuaNeely, 2024-12-09:
**Description**
Our Minio logs bucket is filling up- specifically subdirectories not associated with packet logs.
We have tooling in `plugin.txt` to configure cleanup microservices based on **time and size** of packet logs.
We seem to lack similar functionality for limiting logs for `text_logs` and `tool_logs`, as well as `NOSCOPE` logs.

Minio has time-based bucket lifecycle management but these are less powerful and scalable than something based on filesize, which the cleanup microservices can do.
With bucket sizes constrained by volume mounts (like kubernetes PVCs) it is difficult to correlate time-based cleanup policies with volume size, without massively oversizing volumes.

We would like capability similar to the **CMD_LOG_CYCLE_TIME**, **CMD_LOG_CYCLE_SIZE**, to apply to all logs cosmos writes to the logs bucket.

**Proposed Solution**
Allow configuring the built in cleanup microservice to apply to all content in the logs bucket.
This likely needs to live somewhere outside `plugin.txt`. A global config file? `init.sh`?

**Additional context**
We can configure new minio cleanup policies through the cosmos-init container in the short term, and increase the size of our PVCs.

JoshuaNeely, 2024-12-09:
possibly related to this issue on script-runner log retain time
https://github.com/OpenC3/cosmos/issues/701

jmthomas, 2025-02-14:
closed by #1891 and https://github.com/OpenC3/cosmos-enterprise/pull/217

closed 2025-02-14 by jmthomas