OpenC3/cosmos #1750: Is there documentation for hooking up the monitoring to Datadog?

badrobit, 2024-12-09:
I had checked out [this page in the documentation](https://docs.openc3.com/docs/guides/monitoring), but I wasn't getting many of the answers I expected tas I have seen with other integrations. I was hoping you might have some "unofficial" documentation using [Datadog](https://docs.datadoghq.com/agent/?tab=Linux) or an [open-telemetry](https://opentelemetry.io/) compliant agent to allow for distributed monitoring.

ryanmelt, 2024-12-11:
COSMOS supports open telemetry already (though I haven't tested it for awhile).
You need to set these two environment variables on the containers that you want to use with open telemetry:

OTEL_EXPORTER_OTLP_ENDPOINT=your endpoint url
OPENC3_OTEL=ALL

Note that using open telemetry is not side effect free and hurts performance

badrobit, 2024-12-11:
Would you happen to know what the level of impact is? I fully expected there to be some impact, but I'm curious if any profiling has been done so we could determine whether it is worth deploying.

ryanmelt, 2024-12-11:
Its not horrible, I would say just try it.  
I don't recommend leaving it on all the time though, generally just for debugging things like overall performance.

closed 2024-12-18 by badrobit