OpenC3/cosmos #1752: Any special characters in Template Accessor?

tsjordan-eng, 2024-12-10:
My template uses `#` as a delimiter, but this doesn't seem to work for my configuration.

There's not any documentation on the Template Accessor, but building off of examples I've attempted the following:

Incoming UDP packets:
```
T=2024-12-10T12:20:13.091#D=401#E=1#N=MANET 1 to 2#V=850.0
T=2024-12-10T12:19:25.696#D=3#V= 635.113
T=2024-12-10T12:45:58.180#D=201#C=RF17#E=1#N=proj-RX1#V=No signal
```
T=Time
D=packet type

So I came up with the following TLM file:

```
TELEMETRY PROPSIM IO_STATUS BIG_ENDIAN "Emulation Time"
  ACCESSOR TemplateAccessor
# T=2024-12-10T12:45:58.180#D=201#C=RF17#E=1#N=proj-RX1#V=No signal
  TEMPLATE "T=<TIME_STR>#D=<TYPE>#C=RF<CHANNEL>#E=<E>#N=<IO>#V=<SIG_STATUS>"
  APPEND_ITEM TIME_STR 32 STRING "Packet Time String"
  APPEND_ID_ITEM TYPE 32 UINT 201
  APPEND_ITEM CHANNEL 16 UINT "Channel number (RF#)"
  APPEND_ITEM E 32 UINT "E field?"
  APPEND_ITEM IO 32 STRING "IO Name"
  APPEND_ITEM SIG_STATUS 32 STRING "Status of signal"

TELEMETRY PROPSIM EMULATION_TIME BIG_ENDIAN "Emulation Time"
  ACCESSOR TemplateAccessor
# T=2024-12-10T12:19:25.696#D=3#V= 635.113
  TEMPLATE "T=<TIME_STR>#D=<TYPE>#V=<EMULATION_TIME>"
  APPEND_ITEM TIME_STR 32 STRING "Packet Time String"
  APPEND_ID_ITEM TYPE 32 UINT 3
  APPEND_ITEM EMULATION_TIME 32 FLOAT "Emulation Time"
    UNITS Seconds s

TELEMETRY PROPSIM LINK BIG_ENDIAN "Emulation Time"
  ACCESSOR TemplateAccessor
# T=2024-12-10T12:20:13.091#D=401#E=1#N=MANET 1 to 2#V=850.0
  TEMPLATE "T=<TIME_STR>#D=<TYPE>#E=<E>#N=<LINK>#V=<DOPPLER>"
  APPEND_ITEM TIME_STR 32 STRING "Packet Time String"
  APPEND_ID_ITEM TYPE 32 UINT 401
  APPEND_ITEM E 32 UINT "E field?"
  APPEND_ITEM LINK 32 STRING "Link name"
  APPEND_ITEM DOPPLER 32 FLOAT "Doppler"
    UNITS Hertz Hz
```

However, only one packet gets recognized depending on the order I put it in, the rest come through as `UNKOWN`.

I tried defining the following packet for debugging
```
# TELEMETRY PROPSIM GENERAL_TEMPLATE BIG_ENDIAN "Template"
#   ACCESSOR TemplateAccessor
# # T=2024-12-10T12:20:13.091#D=401#E=1#N=MANET 1 to 2#V=8750.0
#   TEMPLATE "T=<TIME_STR>#D=<TYPE>#<RAW_STRING>"
#   APPEND_ITEM TIME_STR 32 STRING "Packet Time String"
#   APPEND_ITEM TYPE 32 UINT
#   APPEND_ITEM RAW_STRING 1024 STRING "Remaining raw string"
```
but then it gets error. I don't know why it would match the rest of the string into the `TYPE` ITEM.
```
2024-12-10T20:17:34.947972Z ERROR: Decom error: ArgumentError : invalid value for Integer(): "101#C=RF18#E=1#N=prog-TX1"
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/accessors/accessor.rb:123:in `Integer'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/accessors/accessor.rb:123:in `convert_to_type'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/accessors/template_accessor.rb:92:in `block in read_items'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/accessors/template_accessor.rb:85:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/accessors/template_accessor.rb:85:in `read_items'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/packets/structure.rb:157:in `read_items'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/packets/packet.rb:718:in `read_items'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/packets/packet.rb:1258:in `decom'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/models/cvt_model.rb:34:in `build_json_from_packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/topics/telemetry_decom_topic.rb:37:in `block in write_packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/topics/telemetry_decom_topic.rb:29:in `write_packet'
/openc3/lib/openc3/microservices/decom_microservice.rb:123:in `block in decom_packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'
/openc3/lib/openc3/microservices/decom_microservice.rb:87:in `decom_packet'
/openc3/lib/openc3/microservices/decom_microservice.rb:70:in `block (2 levels) in run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:184:in `block (3 levels) in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:182:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:182:in `block (2 levels) in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:181:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:181:in `block in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:110:in `block (2 levels) in with'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:109:in `handle_interrupt'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:109:in `block in with'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:106:in `handle_interrupt'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:106:in `with'
```

Do you have any ideas?

jmthomas, 2024-12-11:
It's fine to use `#` in your template ... the only thing you need to watch out for is the delimiters which is `<>` by default. Your definitions look fine to me. I think the issue might be the identification process. Can you try adding [TLM_UNIQUE_ID_MODE](https://docs.openc3.com/docs/configuration/target#tlm_unique_id_mode) to your target.txt and see if that helps.

tsjordan-eng, 2024-12-11:
Ah, great documentation! I will try out the `*UNIQUE_ID_MODE` tomorrow.

Would the `#` within a string accidentally comment-out the rest of the string? The [File Format docs](https://docs.openc3.com/docs/configuration/format) don't mention comments.

jmthomas, 2024-12-12:
If your `#` is inside quotes then it is part of the string and shouldn't be interpreted as a comment.

tsjordan-eng, 2024-12-12:
The `TLM_UNIQUE_ID_MODE` did the trick, thanks!

closed 2024-12-12 by tsjordan-eng

jmthomas, 2024-12-12:
We have an open issue to auto-detect this:  https://github.com/OpenC3/cosmos/issues/1168