OpenC3/cosmos #1764: Different function calls from one file.

mountTom, 2024-12-16:
I am trying to port over code from COSMOS4 to 5, and I am running into some function call issue. Previously If I wanted to call a specific function from a button press. (Function bar in file foo.rb)

I would add at the top of the screen
`<% 
   require 'foo'
%>`
then for the button call
`BUTTON 'Test' 'bar(arg1, arg2)'`

This would allow me to call multiple different functions in foo.rb.

From what I can tell, this is no longer functional, And I cannot seem to find any alternative that allows me to define the function. runScript() seems to only take a file. Do I have to split all functions to their own files, or is there a way to do this?

ryanmelt, 2024-12-16:
runScripts runs scripts in ScriptRunner.  They need to be files.  They are referenced like "TARGETNAME/procedures/myscript.rb"

mountTom, 2024-12-16:
Got it! Thank you

closed 2024-12-16 by mountTom