OpenC3/cosmos #1774: Passing screen variables into Script Runner scripts

mountTom, 2024-12-17:
I am looking to pass in a screen variable into a Script runner script. This would allow me to be able to set and change the state of widgets in a ruby/python script. 

This was able to be done in COSMOS 4 with the "self" keyword, but that no longer seems to work. Is there an alternative that I could use?

jmthomas, 2024-12-17:
The way you do this in COSMOS 6 is to set environment variables that you pass to your script which the script then reads. We do exactly this in our DEMO in the [commanding](https://github.com/OpenC3/cosmos/blob/e5095c6ca45f0467bffa23b612da7adff8aa9287/openc3-cosmos-init/plugins/packages/openc3-cosmos-demo/targets/INST/screens/commanding.txt#L4) screen and the [checks.rb](https://github.com/OpenC3/cosmos/blob/e5095c6ca45f0467bffa23b612da7adff8aa9287/openc3-cosmos-init/plugins/packages/openc3-cosmos-demo/targets/INST/procedures/checks.rb#L4) script.

closed 2024-12-17 by jmthomas

mountTom, 2024-12-17:
I am looking to SET these widgets, not a GET.  Looking at checks.rb, they are reading the widgets and sending commands. I am looking for reading TLM and setting Widgets

ryanmelt, 2024-12-17:
The main way to pass data to widgets is through telemetry.   You could use set_tlm in the script to set a telemetry point to something that the widget would then read.

Alternatively you'll need to do everything in Javascript in the button code.   You can read telemetry in javascript and then set the widget.  Jason just commented with how to read telemetry in javascript.

mountTom, 2024-12-17:
Understood, Thank you for the help!