OpenC3/cosmos #1777: local_screen in Telemetry Viewer

mountTom, 2024-12-18:
In Telemetry Viewer, I have a button that triggers a ruby script with runScript(). This script does multiple calculations, then opens a local screen for some "local" interaction. This is using the local_screen() function defined in the documents. 

When I run "runScript" with the foreground flag enabled, the script runs correctly in Script runner, and the local screen displays in Script runner. When I set "runScript" to run in the background however, the local screen does NOT display in Telemetry viewer. 

Is there a way to display local screens in Telemetry viewer with ruby?

On a side note, it seems like there is a comma missing in the local screen documentation. 
`local_screen("<Screen Name>" "<Definition>", <X Position (optional)>, <Y Position (optional)>)`

should be &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;⬇

`local_screen("<Screen Name>", "<Definition>", <X Position (optional)>, <Y Position (optional)>)`

jmthomas, 2024-12-19:
@mountTom Thanks for the update on the docs!

If your script is running in the background there isn't anywhere for the local screen to appear so it doesn't really make sense to display a screen.

mountTom, 2024-12-19:
Is there a way to have a local screen in the Telemetry viewer window?

jmthomas, 2024-12-19:
No telemetry viewer is not scriptable ... all the screens are "permanent".

mountTom, 2024-12-19:
Understood, thank you!

closed 2024-12-19 by mountTom